BIPES/BIPES #254: Problème avec les fonctions

Romain888, 2025-04-26:
Hi Jim,

I'm having trouble with functions.
When I put part of my program into a function (simple, no input or output parameters), I get this error message telling me that the function's reference variable is misplaced.
When the program doesn't use functions (a single block column), it works fine.
I don't know how to fix this.
Thanks for your help and have a nice day.

Romain

[image]

[image]

[image]

sk8board, 2025-04-26:
If you are familiar with Python/MicroPython programming, you can see the text form of the code as shown in this screenshot. This will show you the code line number that is referenced in the error message.

[image]

Romain888, 2025-04-27:
Hi,

Yes, I saw that we could see the line numbers affected by the problem.
But I don't see why lines 47 and 87 are reported as errors.
Do you have any ideas on how to resolve this problem?
Thanks.

Romain

[image]

[image]

[image]

jetannenbaum, 2025-04-27:
The problem is due to the use of the variable Liste_WiFi.  BIPES assumed that the variable will be a global variable( Note the declaration  on line 39).  BIPES "helped" by then declaring Liste_WiFi on line 9 as None.  You can fix this by either declaring Liste_WiFi as an empty list on line 9 or by removing the variable from your function.