BIPES/BIPES #258: What is the relation between ui/pylibs and ui/pylibsBlobs?

drucknase, 2025-07-27:
Hello,
what is the relation between the libraries in ui/pylibs and the Files in ui/pylibsBlobs? If I change or create new blocks as described in the documentation do I have to change also code in ui/pylibsBlobs?
Thanks for some explanation.

Wolfgang

drucknase, 2025-07-28:
Hi Jim,
thank you for your fast answer. I'm using for the moment the offline branch, installed locally on my laptop, because I try to use the RC522 RFID reader with the Raspberry Pico but without success. It seems to me, that there is some mismatch between the Python code of the RC522 Block and the used mfrc522.py library. I started to do some debugging but still at the beginning. The call from some library functions from the RF522-Block python code does not match the number of arguments. During debugging I tried to follow the python code so this was the reason to look at ui/pylibs and found then also ui/pylibsBlobs.
What do you mean with 'support files'? The .py library files? In this case I would expect for for eacht file in ui/pylibs a corresponding file in ui/pylibsBlobs, but this is not the case.
When I see clearer why the RC522 is not working, I will open a new issue in this repository.
Thanks Wolfgang

jetannenbaum, 2025-07-28:
Hi Wolfgang,

I took a quick look at the offline branch code.  There is code in the ui/pylibsBlobs library for the mfrc522 device.  So you can ignore the issues that I raised previously.  If there is a mismatch between the generated code and the device library code, the issue can be traced in the generator_stubs.js file.  The block_definitions.js define what functions can be called by the micro controller.

Can you tell me what isn't working?

drucknase, 2025-07-29:
Hi Jim,
from your explanation of the offline branch I understood that the content of ui/pylibs/mfrc522.py should match with the 'translated' code in ui/pylibsBlobs/mfrc522.js. But there is no match. The ui/pylibs/mfrc522.py origin is https://github.com/wendlers/micropython-mfrc522/blob/master/mfrc522.py and from my point of view the origin for the 'translation' in ui/pylibsBlobs/mfrc522.js is https://github.com/kevinmcaleer/pico-rfid/blob/main/mfrc522.py.
Because the to libraries are somehow different, I get the error on the console:
`Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
TypeError: unexpected keyword argument  'spi_id'`

So from my understanding the code generated  for the RC522 Block in ui/core/generator_stubs.js should match for calls to mfrc522.js. But physically the mfrc522.py is downloaded to the Pico.

So for the moment I do not see, which file has to be adapted to which content.
Another point:
if you look at of mfrc522.js  line 182 "    def anticoll(self,anticolN):\n" +  and compare this to /ui/core/generator_stubs.js line 5348   var code = '(' + stat + ',' + tag + ') = rdr.anticoll()\n';  there is a mismatch frm number of parameters to the function call.

I'm sure you will 'see' the problem.

Thank you for your support
Wolfgang

jetannenbaum, 2025-07-29:
Hi Wolgang,

  I see the problem.  There are a few solutions:
1.  Update  the mfrc522.js to match the mfrc522.py code (probably the best
solution)
     a. Then check to ensure the code in generator_stubs.js is correct for
calling the updated code (probably/maybe is)
     b. The example code under wendlers github rebo will need to be
examined, since it isn't set up for the Pico.
2.  Modify the generator_stub.js code to properly call the mfrc522.js code
(not a great solution, since that will cause the generator_stubs.js code to
further diverge from the master version
3.  Modify the mfrc55.js code such that it conforms to the calling
parameters in generator_stubs.js (again, not ideal, since you haven't
solved the problem of two different versions of the mfrc55 code).

It does look like you will need to add a bit of code to make either
solution work, since the reading and writing for access is written
differently.

Let me know what you want to do, and I'll add comments as you go.

 Jim

jetannenbaum, 2025-07-30:
Hi Wolfgang,

   I looked at the mfrc522.py file.  They used 0x93, which corresponds to MFRC522.PICC_ANTICOLL1.  Give that a try.

Jim

drucknase, 2025-07-30:
Hi Jim,
thanks a lot! It works. Because I did not use all RF522 blocks up to now, I will check the other blocks also in a sketch.

Wolfgang